hyye/lio-mapping #2: Hello, why is rviz no point cloud display?

hunkyu, 2019-04-30:
[image]
[image]
I am using fast1.bag but no rviz has no point cloud display

hyye, 2019-04-30:
Hi @hunkyu, can you check if you build with the flag `-DCMAKE_BUILD_TYPE=Release`? If not, the runtime cannot be ensured and the estimator might not be initialized well. This debug info should not appear with release compilation. The default flag will be added in the `CMakeLists.txt` the future release.

closed 2019-05-10 by hyye

yangjiongfeng, 2019-07-04:
I have the same problem as you.
Have you solved the problem yet?

RalphCodesTheInternet, 2021-09-02:
Same problem. Any solutions yet?