hyye/lio-mapping #5: Some doubts about Supplementary Material partE

icameling, 2019-05-27:


hyye, 2019-05-27:
For any quaternion, there is a left-hand side multiplication equivalent to a right-hand side multiplication. If the perturbation is only on the yaw angle, these two _delta qs_ will be equal.

In the meantime, the estimated orientation in LIO is close to the ground-truth except for the yaw angle, which is a property of the inertial fusion.

icameling, 2019-05-27:
Thanks for your quick reply. I am still confused about this equation.

Is _**delta_L^W**_  a normal perturbation  quaternion? 

Why _**q_L^W**_ left multiplication _**delat qz**_ equal to _**q_L^W**_ right multiplication a normal perturbation quaternion _**delta q_L^W**_?

By the way, you say  if the perturbation is only on the yaw angle, these two delta qs will be equal, Could you explain in detail? 
Hope for your reply. Thanks a lot!

KitKat7, 2019-05-27:
1. It’s a normal perturbation on the right.
2. Since we want to find a constrained left perturbation which is close to a right perturbation.
3. My previous reply is not a good explanation. The yaw-major perturbation (drift) on the global pose can be corrected by a left perturbation.

hyye, 2019-05-27:
1. It’s a normal perturbation on the right.
2. Since we want to find a constrained left perturbation which is close to a right perturbation.
3. My previous reply is not a good explanation. The yaw-major perturbation (drift) on the global pose can be corrected by a left perturbation.

icameling, 2019-05-28:
So in LIO_Mapping, you observed the global pose of the yaw angle is not as accurate as the other 2 dimensions, then the rotation-constrationed maping is applyed to solve this problem. However, this situation may not be common. Our NDT-based lidar mapping suffers from the pitch-major error situation. Here is a part of the result of our NDT-based mapping. May be I misunderstand your meaning, or our mapping algorithm is not as perfect as yours.
[image]

And about the Eqution.12 in supplementary material, could I rewrite the eqution like the second way? How do you come up with this formula? I can't find material on this aspect of quaternion.
[image]

hyye, 2019-05-28:
Since roll and pitch angles are observable in LIO (with inertial fusion), the accumulated error is in yaw angle major. This is different from your NDT-Lidar-only method.

The rotational error of the odometry is majorly in the global yaw angle (unobservable). Thus, I choose to correct the global yaw to make it align with the map. Intuitively, it should be on the left (it can be considered as “\delta q_W^{W’} * q_L^{W’}”). Of course, you can rewrite the equation, but the physical meaning may change and I haven’t try in that direction.

icameling, 2019-05-29:
Thank you for your detailed explanation.:smile:

closed 2019-05-29 by icameling