hyye/lio-mapping #6: some doubts about process died

yuqingyang888, 2019-05-27:
I was playing a dataset recorded by myself. In the first 20seconds, it ran well. However, after that, something got wrong, and I didn't know how to fix it.
The dataset was recorded outside door, and when I used "test_indoor.launch" and "test_outdoor.launch". The mistake was the same.

[image]

hyye, 2019-05-27:
Could you please provide more detailed information about the crash? 

What is your IMU output rates? Did the system begin to drift before the crash?

yuqingyang888, 2019-05-27:
The output rate is 200Hz.
The system indeed began to drift. The picture is a comparison between loam and lio. The LOAM seems operate normally.
[image]
[image]

hyye, 2019-05-27:
If it is planar motion, the current IMU initialization needs some rotation; and in the planar case, the extrinsic parameter estimation can be degenerate.

Perhaps you may try with outdoor config without initial guess of extrinsic parameters (set the flag in config file to 2, and the translational part of extrinsic parameters close to your sensor configuration).

yuqingyang888, 2019-05-27:
I would try to solve it tomorrow. 
And I have another problem that there are two files where I could set the parameters.They are "indoor_test_config.yaml" and "Estimator.h". I should set the parameters in indoor_test_config.yaml. Is that correct?

hyye, 2019-05-27:
You’re right, in the yaml file.

hyye, 2019-07-06:
See more at https://github.com/hyye/lio-mapping/issues/11#issuecomment-508896635.

closed 2019-07-06 by hyye