hyye/lio-mapping #7: Is the raw point cloud de-skewing be implemented？

daijicheng, 2019-05-30:
In the paper, the raw point cloud from lidar sensor is de-skewed before feature extraction, but in the code PointProcessor.cc I can not find this part, did I miss something?

hyye, 2019-05-30:
In my implementation, the linear de-skewing is done by `TransformToEnd` in `src/point_processor/PointOdometry.cc` or `src/imu_processor/Estimator.cc`.

daijicheng, 2019-05-30:
Thanks for your answer. I think extracting feature in the distorted point cloud will have negative effect in the quality of feature points, is this effect serious?

hyye, 2019-05-30:
I don't think it will have serious effects on the feature extraction since the timestamps of consecutive points in one scan are very close. Thus, most of the features should still be preserved, except the points between the end and beginning of one scan.

daijicheng, 2019-05-30:
For the feature only rely on local points, this makes sense, thank you.

closed 2019-05-30 by daijicheng