hyye/lio-mapping #8: hardware time synchronization is not required?

tuandle, 2019-06-08:
Hi @hyye,
Thank you for releasing the source code. 
I have a question about time synchronization between lidar and IMU. My understanding is that for tightly couple approach, you need to have a precise timing of measurement messages. 
If I understand the code correctly, in [this function](https://github.com/hyye/lio-mapping/blob/972568a8d912330992f47d94c341d0a8dce652c6/src/imu_processor/MeasurementManager.cc#L54) from `src/imu_processor//MeasurementManager.cc`, you simply compare the timestamp of IMU and lidar, which I think they are timestamped on arrival, for buffering.
Is it one of the reasons why you need such a high output rate (at 400 Hz) for IMU?

hyye, 2019-06-08:
Hi @tuandle, actually it is not necessary to use an IMU with 400 Hz, but required for relatively higher than lidar’s.

I tested IMUs with stable time stamp and rate typically greater than 100 Hz. Our algorithm can still work.

Time synchronization is indeed an important fact. It is one of the reasons to use IMU with high output rate to reduce its effects. You may try PPS port on lidar to synchronize with IMU, if you want.

tuandle, 2019-06-10:
Thanks for your clarification!

closed 2019-06-10 by tuandle