hyye/lio-mapping #9: Is there something wrong with me?

FitTiggo, 2019-06-25:
[image]
I followed the steps you mentioned to run the data “fast1.bag"  normally, and then  conversion completed the point cloud used "lio_save_bag_to_pcd" with a record bag, The result is like the picture above.  is there something wrong with me?

hyye, 2019-06-25:
Hi @FitTiggo, I think this can be caused by using wrong topic names to build a pcd file.

Could you try with `lio_save_bag_to_pcd --input_filename "[bag_name].bag" --output_filename "[pcd_name].pcd" --odom_name "/lio_map_builder/aft_mapped_to_init" --laser_name "/lio_map_builder/cloud_registered"`?

BTW, it will be better to skip first several PointCloud2 messages, which might be unstable.

FitTiggo, 2019-06-25:
Thank you for your reply so quickly. I ran the data again according to your instructions.The picture below shows the result of cutting off the ground and ceiling. This result is very good, but is it normal?I can see the obvious noise of the wall and the thickness of 20cm.

[image]

FitTiggo, 2019-06-25:
This is the cropped point cloud @hyye 
[project.zip](https://github.com/hyye/lio-mapping/files/3324914/project.zip)

hyye, 2019-06-25:
For fast bags, these were typical results under extreme conditions, since we aimed at evaluating the robustness under fast motion.

If better mapping results are needed, we recommend moderate motion. I tried the slow1 bag; the error of the plane is around 0.1-0.05cm (not accurately measured), which is much smaller.

FitTiggo, 2019-06-25:
Good work!  Thanks for a lot.

closed 2019-06-25 by FitTiggo

zorosmith, 2019-09-07:
Hello, I try to save the map using your command, but I meet the error as following,
_F0907 21:27:52.891425 15780 save_bag_to_pcd.cc:162] no points saved
*** Check failure stack trace: ***_

I run the command in 2 ways, while running  `roslaunch lio test_indoor.launch`,  `roslaunch lio map_4D_indoor.launch` and  `rosbag play unorganized_02.bag`.
1. path::~/ros_ws/devel/lib/lio $ 
`./lio_save_bag_to_pcd --input_filename "/home/gordon/ros_ws/src/LeGO-LOAM/LeGO-LOAM/data/0802exp/unorganized_02.bag" --output_filename "111.pcd" --odom_name "/lio_map_builder/aft_mapped_to_init" --laser_name "/lio_map_builder/cloud_registered"`

2. `rosrun lio lio_save_bag_to_pcd --input_filename "/home/gordon/ros_ws/src/LeGO-LOAM/LeGO-LOAM/data/0802exp/unorganized_02.bag" --output_filename "111.pcd" --odom_name "/lio_map_builder/aft_mapped_to_init" --laser_name "/lio_map_builder/cloud_registered"`

Both commands report error.
Could you give me some help?
Thank you!

hyye, 2019-09-09:
Hi @zorosmith, I think it might be related to the topic name of the point cloud.
If there are no points accessed, this error will appear.

zorosmith, 2019-09-09:
hello @hyye , finally I make sense how to use save bag_to_pcd.cc . 
1. run lio_mapping and rosbag play raw_data.bag.
2. record "/lio_map_builder/cloud_registered" and "/lio_map_builder/aft_mapped_to_init" in a new bag, lio.bag .
3. rosrun lio lio_save_bag_to_pcd --input_filename "lio.bag" --output_filename "./111.pcd" --odom_name "/lio_map_builder/aft_mapped_to_init" --laser_name "/lio_map_builder/cloud_registered"

HTLife, 2021-03-18:
Easier way could be
```
roslaunch lio test_indoor.launch
```
```
roslaunch lio map_4D_indoor.launch
```
```
mkdir lio_pcd
cd lio_pcd
# you can start this line after the rosbag play to skip some frame which is not properly registered
rosrun pcl_ros pointcloud_to_pcd _binary:=true input:=/lio_map_builder/cloud_registered
```
```
rosbag play your_bag.bag
```
To combine the pcd files in lio_pcd folder
```
pcl_concatenate_points_pcd *.pcd
```
You can get the combined file called `output.pcd`

In this way, you don't need that bag to pcd node anymore.