hyye/lio-mapping #10: No map is displayed in rviz

yangjiongfeng, 2019-07-03:
hello 
i have been run 
1.roslaunch lio test_indoor.launch 
2.roslaunch lio map_4D_indoor.launch 
3.rosbag play fast1.bag
but ,No map is displayed in rviz
What's wrong with me?

hyye, 2019-07-03:
I guess it could related to the communication problem in `ROS`. Could you close all your launches and start `roscore` first, before you start other launches and `rosbag`? 
`rostopic` or `rqt_graph` commands can also help you debug with this problem.

BTW, please make sure the code is built with `Release` option.

yangjiongfeng, 2019-07-04:
I have been restart many times，
But as a result, also no map was displayed on rviz 。
What's your WeChat or Email  number?

hyye, 2019-07-04:
Can you make sure if you build the code with **`Release`** option? As I replied here https://github.com/hyye/lio-mapping/issues/2#issuecomment-488046749.

hyye, 2019-07-04:
Try to use `catkin build -DCMAKE_BUILD_TYPE=Release lio` or `catkin_make -DCMAKE_BUILD_TYPE=Release` to rebuild the project.

yangjiongfeng, 2019-07-04:
Thank you！
when i try to use catkin build -DCMAKE_BUILD_TYPE=Release ，I've solved the problem 。
why you project need use catkin build -DCMAKE_BUILD_TYPE=Release but not catkin_make?

hyye, 2019-07-04:
Glad to help!
It is not `catkin build` or `catkin_make` making differences, but the `CMAKE_BUILD_TYPE`.
Typically, `catkin build` or `catkin_make` will not trigger the `Release` flag.

yangjiongfeng, 2019-07-04:
i have build map use you code.
now,i run rosrun rqt_tf_tree rqt_tf_tree,can show de tf_tree
[image]
and then,i run rqt_graph
show 
[image]
only /local/full_point between lio_estimator and lio_map_builder,
but you code have four subscribe
[image]

why build the map only use one subscribe？

hyye, 2019-07-04:
It should be four instead of only one, if it run properly.
Maybe some of the  programs (nodes) are exited.

closed 2019-07-06 by hyye