hyye/lio-mapping #11: Performance in KITTI or velodyne-HDL64e ?

lshGame, 2019-07-05:
Hello, big thanks for sharing this code.
I made a ros bag from KITTI raw data(2011_09_30_drive_0027_extract). This bag includes cam0, 100Hz imu and velodyne-HDL64e. I have tested this bag with VINS-Mono and lio-mapping(64_scans_test.launch), the result is good.
But the result of lio-mapping(test_outdoor.launch & map_4D_indoor.launch) is bad. lio-mapping is able to initialize but can't reach high level of accuracy in velodyne-HDL64e.
[image]

hyye, 2019-07-06:
From our experience on the outdoor planar motion tests, two main issues can be the prior extrinsic parameters (if no auto rotation calibration is applied) and initialization.
Sometimes, initialization may suffer from “bad motion”, bad extrinsic parameters or wrong data, yet it still claims to be successful.
For further optimization, extrinsic parameters are important to the initial states. Straight / planar motion is not friendly to the auto extrinsic parameter estimation.
Maybe you can share the tools to build the KITTI data. I can have a try when I have time.

weisongwen, 2019-07-06:
@, hi haoyang, Thanks for sharing your nice work. The prior factor is the prior for extrinsic parameters?

Best,
Welson,

hyye, 2019-07-08:
@weisongwen, yes it is. This is only needed when translation parts of the extrinsic parameters cannot be well constrained.

weisongwen, 2019-07-08:
Hi @hyye , quite clear, Thanks.

KitKat7, 2019-07-23:
@lshGame, I tested the KITTI 00 sequence with the commit 8485a4d9c359b7e486ad019a10815ecb8fd1257f. It should work.
You may try to run `test_outdoor_64.launch` and `map_4D_indoor.launch` to see the results w/o and w/ mapping.
The estimator will have some drift at the beginning; but after several seconds, it converges.

hyye, 2019-07-23:
@lshGame, I tested the KITTI 00 sequence with the commit 8485a4d. It should work.
You may try to run test_outdoor_64.launch and map_4D_indoor.launch to see the results w/o and w/ mapping.
The estimator will have some drift at the beginning; but after several seconds, it converges.

closed 2019-08-05 by KitKat7