hyye/lio-mapping #12: how to modify indoor_test_config.yaml

yangjiongfeng, 2019-07-08:
Hi anthor，
My robot moves horizontally indoors.
You don't known Extrinsic paramete，so in  you indoor_test_config.yaml ,Your estimate_extrinsic is 2.
now i have Extrinsic parameter between IMU and Camera,How to modify my parameters，
Please tell me which  parameters and how I need to modify. 
Thank you

hyye, 2019-07-09:
Hi @yangjiongfeng, you can set `estimate_extrinsic` to 1, and set your `extrinsic_rotation` and `extrinsic_translation` (from the IMU frame to the lidar frame).
It is recommended to estimate the extrinsic parameters first and make sure the system work properly; then set it as a prior value in further use.

yangjiongfeng, 2019-07-12:
hello anthor，
How do you save the  map as pcd file?

hyye, 2019-07-19:
Hi @yangjiongfeng, to save as pcd file, you can add storing code after the loop of `ros::ok`, like https://github.com/hyye/lio-mapping/blob/972568a8d912330992f47d94c341d0a8dce652c6/src/map_builder_node.cc#L77 or record the messages in a bag and then use the tool `src/save_bag_to_pcd.cc`, referring to https://github.com/hyye/lio-mapping/issues/9#issuecomment-505382679.

llu0120, 2019-08-20:
Hi author,
Where should I use the tool `lio_save_bag_to_pcd --input_filename "[bag_name].bag" --output_filename "[pcd_name].pcd" --odom_name "/lio_map_builder/aft_mapped_to_init" --laser_name "/lio_map_builder/cloud_registered"`, I run this in the terminal but it failed like `lio_save_bag_to_pcd: command not found`, any suggestions?

llu0120, 2019-08-26:
HI author, 
I saw there is a file of `save_bag_to_pcd`. Per the discussion before, I saw I might need to run the command `lio_save_bag_to_pcd --input_filename "[bag_name].bag" --output_filename "[pcd_name].pcd" --odom_name "/lio_map_builder/aft_mapped_to_init" --laser_name "/lio_map_builder/cloud_registered"`, however, I don't know where to run this and should the bag be a recorded bag by `rosbag record /topics_name`?  And can you give me an example to run the `lio_save_bag_to_pcd` command?