hyye/lio-mapping #13: Process died after catkin_make -DCMAKE_BUILD_TYPE=Release and source devel/setup.bash

llu0120, 2019-07-08:
I run the below command 
`roslaunch lio test_indoor.launch &` and the process is died like below:
[image]
and after that I run the below command 
`roslaunch lio map_4D_indoor.launch` and the process is died like below: 
[image]

Do anyone know how to solve this to make the code run?

hyye, 2019-07-09:
Hi @llu0120, I have no idea about the outputs you posted here.
What dataset do you use? What is your platform?
If you did not play any `rosbag` and the system just died, it might be related to some system-wise libraries.
If possible, could you run the program within docker?

llu0120, 2019-07-09:
Hi author,
Thank you for your reply!
I didn’t play any bag and the system just died after I do those roslaunch commands. 
My platform is Ubuntu 16.04 with ros kinetic, and I will not use docker. Any reference of the version of each library? Or any other solution?

hyye, 2019-07-10:
Maybe rebuilding with `Debug` flag and using `gdb` to see when the program died is a fast way.

For reference, the libraries for `Ubuntu 18.04` are the same as the ones in the default docker image.
- `ROS`: 1.14.3
- `OpenCV`: 3.2.0
- `Ceres`: 1.14.0
- `PCL`: 1.8.1

llu0120, 2019-07-10:
I’m using 16.04 instead of 18.04, are the references still right?

ychenco, 2019-07-10:
I'm also using 16.04. It works fine in my destop. The configurations are:

- ROS: kinetic
- OpenCV: 3.3.1
- Ceres: 1.14.0
- PCL: 1.7.2

closed 2019-07-12 by llu0120