hyye/lio-mapping #14: Playing my data set after 20 secs lio_processor died

llu0120, 2019-07-12:
Hi author, 
I run the open source on my own data, 
first I run `roslaunch lio test_indoor.launch &` and then run `roslaunch lio map_4D_indoor.launch &`. After these command, I play my bag by adding `--clock` at the end and after running for 20 sec, the process is died like below. 
[image]
However, if I don't add `--clock ` at the end, the process will not died in the whole bag, but the terminal of `roslaunch lio test_indoor.launch &` didn't show anything and also nothing shown in rviz as well. Any suggestions for this bug? Thank you!

hyye, 2019-07-12:
Did you use self-collected data?

I didn't meet this problem. I don't think it is a common "bug". With only error information from the terminal, it is tough to say where the problem is.
As I said before, using gdb to find out where the program dies can help us to locate the problem.

llu0120, 2019-07-15:
Thanks for your reply! 
I use gdb and the result looks like below:
[image]
[image]

hyye, 2019-07-15:
Does the provided dataset work?
It seems some problems with `PointProcess`. Is there any NAN data in your point cloud? If yes, maybe try to filter the NAN data.

But, I believe it is a problem with some libraries. My suggestion is to clean and rebuild the project first.

BTW, what is your `PCL`, `Eigen` and `gcc` version?

llu0120, 2019-07-15:
The provided dataset printed out the messages without errors like below 
[image]
However, Rviz didn't show any topics visualization like below 
[image]
and my `PCL `version is 1.8.0, `Eigen` is 3.2.92, `gcc` is 5.4.0.

hyye, 2019-07-15:
There are no results since it was not built with `Release` option. See https://github.com/hyye/lio-mapping/issues/2#issuecomment-508306469 and https://github.com/hyye/lio-mapping/issues/10#issuecomment-508308072.

llu0120, 2019-07-15:
But if I build with Release option, the process estimator will died like below
[image]

hyye, 2019-07-15:
Then, I am not sure where the problem is, potentially related to some libraries. My final recommendation is to use `docker` or another computer to see if the same problem exists.

llu0120, 2019-07-15:
Thanks for the suggestion! However, because of being new to using docker, are there any instruction to setup the environment?

hyye, 2019-07-15:
I suggest to follow [docker docs](https://docs.docker.com/install/linux/docker-ce/ubuntu/) and [nvidia-docker wiki](https://github.com/nvidia/nvidia-docker/wiki/Installation-(version-2.0)) to setup the basic docker environment.

Then, use the commands I provided in the [README](https://github.com/hyye/lio-mapping#docker) to build and run docker. I hope this can help you.

llu0120, 2019-07-15:
When I tried the command `sudo apt-get install nvidia-docker2`, some problem happened like below,
[image]
I Googled it but still couldn't fix it, can you help me? Thank you!

llu0120, 2019-07-15:
Now my docker version is `Docker version 18.09.7, build 2d0083d` and I have install nvidia-docker2 by `sudo apt-get install -y nvidia-docker2 ` and  `pkill -SIGHUP dockerd` successfully. What is the next step, how to run your readme command `docker/build_docker.sh`, I run like below but it didn't pass. 
[image]

hyye, 2019-07-15:
Well, you need to run the command in the directory `lio-mapping`.

llu0120, 2019-07-16:
Hi thanks for your instruction, I successfully run your provided dataset with docker. But now I try to plugin my dataset, but the process died again like below. :(
[image]

hyye, 2019-07-16:
Does the provided dataset work?
Maybe try to check the `NAN` values or incompatible data first. The default rate of 3D lidar is 10Hz.

In docker, you can still use some commands like `docker exec` to enable `gdb`. It seems a problem related to memory corruption.

llu0120, 2019-07-18:
Yeah it works. But my dataset doesn't, does this mean I need to change the default rate of the 3D lidar compatible to my dataset. I enable the `removeNaNFromPointCloud` first to avoid nan values.

hyye, 2019-07-19:
If it is standard 3D lidar, like Velodyne, with default settings and drivers. The data should be OK.
It should be a sweep of points (> ten thousand points) in one point cloud with update rate around 10Hz.

llu0120, 2019-07-22:
Still stuck at this memory corruption problem. It is a standard 3D lidar and I found the difference between my bag and the bags you provide that my `/imu/data` is with frequency 50 Hz but your bags are 400 Hz. Will it be a reason? The ROS messages are normal for the first iteration passing `/laser_cloud_sharp` at 10 hz for two times and passing `/local/surf_points` at 5 hz for one time, but after that the memory corruption problem happens.
Below is another comparison of the `rqt_graphs`.

[image]
[image]

llu0120, 2019-07-23:
I use `gdb` to debug and found that when confronting this line `curvature_idx_pairs_.resize(region_size);` in the `PrepareSubregion` function many times, it will crash with the `memory corruption` error. Any idea why this happened?

KitKat7, 2019-07-23:
Can you check the `region_size`? Is it too large or invalid?

hyye, 2019-07-23:
Can you check the `region_size`? Is it too large or invalid?

llu0120, 2019-07-23:
Thanks for your reply. The value of the `region_size` were around 200 for the first few scan, but then it has suddenly increase to 633 (looks too large), which also corrupt at this point. As the `region_size` is define by `ep` `sp` and those two are define by `scan_size`, but the `scan_size` are the data grab from my lidar, so I don't think I can change that. Any suggestion?

hyye, 2019-07-23:
BTW, what is your platform? Does it have enough memory to run the program?
One suggestion is to add some [exception handling](https://docs.microsoft.com/en-us/cpp/cpp/try-throw-and-catch-statements-cpp?view=vs-2019) to enclose the line causing the problem.

But it can still relate to the driver or data.
I haven't tried Pandar. The current version does not directly support lidars that are not with 16, 32, or 64 scans.

llu0120, 2019-07-24:
Thanks Author, 
I have solve this problem for changing the computation of `ElevationToRing` since my lidar is different than yours.

closed 2019-07-24 by llu0120