hyye/lio-mapping #15: can not use imu data

ZivKidd, 2019-07-17:
First of all, your job is really brilliant.
I am new for slam, and I have test your method by our own data collected by hesai lidar and stim 300 imu, but the rqt_gragh result shows that only the lidar data was used, so what should I do to use both lidar and imu data?
Thank you very much. My launch file is below.

<launch>

    <arg name="rviz" default="true" />

    <node pkg="lio" type="lio_processor_node" name="lio_processor" output="screen">
        <param name="sensor_type" type="int" value="40" />
        <remap from="/velodyne_points" to="/pandar_points"/>
    </node>
    <node pkg="lio" type="lio_odometry_node" name="lio_odometry" output="screen" />
    <node pkg="lio" type="lio_mapping_node" name="lio_mapping" output="screen" />

    <group if="$(arg rviz)">
        <node launch-prefix="nice" pkg="rviz" type="rviz" name="rviz" args="-d $(find lio)/rviz_cfg/lio_baseline.rviz" />
    </group>

</launch>

hyye, 2019-07-17:
Hi @ZivKidd, what is your IMU topic? I think it is required to remap your own IMU topic to `/imu/data`.

In the current version, 40 line lidar is not supported. If your data has `ring` information, then you may refer to `sensor_type == 320`, which is used for 32 line lidar with `ring` information, in order to modify the source code of `lio_process`.

ZivKidd, 2019-07-18:
Thanks for your reply.
My IMU topic is  /imu/data.

topics:      /imu/data        213017 msgs    : sensor_msgs/Imu        
             /pandar_points     2235 msgs    : sensor_msgs/PointCloud2

And I have modified the source code to satisfy my lidar.

    PointProcessor processor; // Default sensor_type is 16
    if (sensor_type == 32) {
        processor = PointProcessor(-30.67f, 10.67f, 32);
    } else if (sensor_type == 64) {
        processor = PointProcessor(-24.9f, 2, 64);
    } else if (sensor_type == 320) {
        processor = PointProcessor(-25, 15, 32, true);
    } else if (sensor_type == 40) {
        processor = PointProcessor(-16, 7, 40, true);
    }

So which changes should I make to use the imu data?
Thank you.

ZivKidd, 2019-07-18:
Is lio_estimator_node the node I need to use my imu data? So I need to calibrate my imu and lidar first, and when I get the extrinsic parameters, lio_estimator_node will add my imu data for the mapping?

ZivKidd, 2019-07-18:
when the program calibrate my imu and lidar, how can I let it publish the extrinsic parameters?

hyye, 2019-07-19:
Hi @ZivKidd , there is a topic `extrinsic_lb`.

shul-aowu, 2019-07-22:
hello， @hyye ,  i have a question .   The extrinsic_lb is the optimize parameter, when imu finished initialization ,  i see the extrinsic_lb published  will not change .

hyye, 2019-07-22:
Hi @shul-aowu, if there is no input, it will not change.
I checked `fast1.bag`. The extrinsic parameter indeed changes all the time.

shul-aowu, 2019-07-22:
Thanks for replying so  quickly. I modify a para and cause the  problem.
and another question: my imu installation has a transform(oritation and translation) to the lidar , and I use the out_door.yaml  to test my data (outdoor) , set the  extrinsic_rotation .  But using my data ,I found the z axis's drift  , what may cause the problem ? @hyye

hyye, 2019-07-22:
@shul-aowu, if it's planar motion, the extrinsic parameter of z-axis will drift because of no enough observation (constraints). The other states can still be well estimated with this drift.

It will be better to open a new issue instead of replying in a closed issue if you have further questions.