hyye/lio-mapping #17: imu excitation is not enouth

pxc2017, 2019-07-22:
Hello，hyye.First,thanks for your open work,that is so amazing.im green in SLAM. I just confused about why “imu excitation is not enouth”appeared while using my own dataset,and about one minute later,with the imu initialized successfully,the map occurred but the datas in that one minute has lost.Is frequency of the imu not high enough?I used the MTi Xsens 100Hz,RS-16 10Hz
BTW,should it be "enough",i also saw "enouth"in the source code.
thank you very much!

hyye, 2019-07-22:
It's a typo in the code.
`The excitation is not enough` means that the current motion is not enough for IMU to be initialized, like constant velocity, no rotation. This behavior ensures the system has a good initial value for further optimization.

BTW, if you use `Xsens`, it might be better be set to a higher frequency.

pxc2017, 2019-07-22:
really appreciate for your answer,i will make a try.

llu0120, 2019-07-26:
Hi, I also confront this problem, that my IMU frequency is 50 Hz, will this lead the not initialized IMU?

hyye, 2019-07-27:
@llu0120, see more discussions in https://github.com/hyye/lio-mapping/issues/22.

closed 2019-07-27 by hyye