hyye/lio-mapping #19: Compatibility with Google Cartographer datasets

andre-nguyen, 2019-07-22:
Hi @hyye ,

How would you adapt lio-mapping to work with the google cartographer datasets? I see they converted each lidar packets to a pointcloud while you use the accumulated clouds from the velodyne driver. Other than reassembling clouds, are there any parameters that neet to be adapted?

Thanks,

hyye, 2019-07-23:
Hi @andre-nguyen, I haven't tried to reassemble the clouds from the cartographer datasets yet.

I think in reassembling, we need to make the cloud as a full sweep and set the timestamp of the cloud as the middle point's. The [Velodyne's ROS driver](https://github.com/ros-drivers/velodyne) may be a useful reference.

andre-nguyen, 2019-07-23:
Thanks, I stamped with the end of the cloud and it seemed to work. Will look into which time stamp works best. Did you use the fully assembled clouds from the ros driver?

andre-nguyen, 2019-07-23:
https://github.com/ros-drivers/velodyne/blob/master/velodyne_driver/src/driver/driver.cc#L252

Looks like they stamp with the last timestamp. I find it interesting that lio-mapping works so well when the timing isn't particularly tight.

hyye, 2019-07-23:
Glad to hear it works!
I used the fully assembled one from the ros driver.
I got it wrong; It should be the end of the cloud. I transformed the points to the middle time in the code.

andre-nguyen, 2019-07-23:
Unfortunately, it crashes on some of the datasets due to some kind of kdtree error. It is unclear to me why. However, I have my own dataset that is formatted like the cartographer ones and lio seems to work on mine.

closed 2019-07-23 by andre-nguyen