hyye/lio-mapping #22: Extrinsic Calibration Failed and imu not initialized?

llu0120, 2019-07-25:
It seems that running the data you provide will have `Extrinsic Calibration successful` and `imu initialized`. However, now I plugin my data and the bag can play normally to the end, but didn't show anything on Rviz and always show Extrinsic Calibration Failed in the terminal. Does this mean the result didn't converge? How to deal with this problem or what might cause them happened?

llu0120, 2019-07-26:
Hi author,
After tuned the extrinsic parameters, I confronted `imu excitation is not enouth` and I found that your provided IMU frequency is about 400 Hz, but mine's is 50 Hz. Will this be a problem?

hyye, 2019-07-26:
Typically, 50Hz is not enough for the system to run smoothly.

llu0120, 2019-07-26:
May I know the reason that why this frequency doesn't work?

hyye, 2019-07-26:
One issue is the timestamp alignment; another issue is the integration error will be large when the time interval is large.
I don't suggest using a low-frequency IMU.

llu0120, 2019-07-26:
So is there a suggestion of frequency? Or what is the threshold of this frequency?

hyye, 2019-07-26:
My suggestion is at least `>=100Hz`, recommended `>=200Hz`.
`imu excitation is not enough` (the typo is fixed) means no good initialization can be obtained. Thus, there is no further optimization; otherwise, only the wrong results will be produced.

llu0120, 2019-07-26:
Thanks for your reply, but due to my data the imu frequency cannot change. Are there any parameters that can adjust?

hyye, 2019-07-26:
It is not recommended to use a low-frequency IMU. But you may try to moderately move (rotate) the sensor pair to make the extrinsic calibration success. Then fill in the calibrated parameters and set the `estimate_extrinsic` in the config file to 1.

llu0120, 2019-07-27:
Thanks for your suggestion! By the way what is the function `ApproximateGravity` for? Because I make the return always into `true` then the imu can be initialized. (My IMU data is already processed by deducting g_norm in the z axis of accleration.)

hyye, 2019-07-27:
That function is used to estimate the gravity vector in the local frame.
If your IMU data provides free acceleration, the code cannot be directly used. In the current version, the inputs should be raw IMU data without subtracting gravity.

llu0120, 2019-07-31:
Another question, what is the real usage of the rostopic /compact_data, can I turn it off?

hyye, 2019-08-01:
To turn off IMU data, you can set `imu_factor` to `0` in the config yaml . Then, there will be no optimization at all.
`compact_data` is used for storing all the feature points, odom, etc. and transmit all of them once by one message.

closed 2019-08-20 by llu0120