hyye/lio-mapping #23: Some questions about CircularBuffer, enable deskew and Estimator::BuildLocalMap

zhaostu4, 2019-07-26:
hello ,hyye! Thanks for your brilliant work.
Now I am trying to understand your code and have some questions.
1)	the function Estimator::BuildLocalMap maintaining the local map, and solves the pose of the last frame. In an outdoor environment, the flag of keep_Feature is false, the function Estimator::CalculateFeatures will accumulate all features from pivot_idx to window_size, and these features will be used to solve the pose of the lase frame. however, these features from pivot_idx to windows_size -1 are not associated with the pose of the last frame, the points of these features can't transform to the pivot_idx frame though the pose of the last frame.
2)	Why not enable deskew in PointOdometry with imu, intuitively it's better.
3)	the class of CircularBuffer may be a small bug, its index is out of order, which means that the function Estimator::SlideWindow may give the next frame an abnormal initial value. This is a simple example
code:
    int loo = 8;
    CircularBuffer<int> buf{loo};
    for (int i=0; i<20; i++)
    {
        buf.push(i);
        std::cout << "||\tindex: "<< i << "\tget: "<< buf[i]<< std::endl;
    }
Result:
CircularBuffer: 0	||	index: 0	get: 0
CircularBuffer: 0	1	||	index: 1	get: 1
CircularBuffer: 0	1	2	||	index: 2	get: 2
CircularBuffer: 0	1	2	3	||	index: 3	get: 3
CircularBuffer: 0	1	2	3	4	||	index: 4	get: 4
CircularBuffer: 0	1	2	3	4	5	||	index: 5	get: 5
CircularBuffer: 0	1	2	3	4	5	6	||	index: 6	get: 6
CircularBuffer: 0	1	2	3	4	5	6	7	||	index: 7	get: 7
CircularBuffer: 8	1	2	3	4	5	6	7	||	index: 8	get: 1
CircularBuffer: 8	9	2	3	4	5	6	7	||	index: 9	get: 3
CircularBuffer: 8	9	10	3	4	5	6	7	||	index: 10	get: 5
CircularBuffer: 8	9	10	11	4	5	6	7	||	index: 11	get: 7
CircularBuffer: 8	9	10	11	12	5	6	7	||	index: 12	get: 9
CircularBuffer: 8	9	10	11	12	13	6	7	||	index: 13	get: 11
CircularBuffer: 8	9	10	11	12	13	14	7	||	index: 14	get: 13
CircularBuffer: 8	9	10	11	12	13	14	15	||	index: 15	get: 15
CircularBuffer: 16	9	10	11	12	13	14	15	||	index: 16	get: 9
CircularBuffer: 16	17	10	11	12	13	14	15	||	index: 17	get: 11
CircularBuffer: 16	17	18	11	12	13	14	15	||	index: 18	get: 13
CircularBuffer: 16	17	18	19	12	13	14	15	||	index: 19	get: 15

hyye, 2019-07-26:
1. For speed up.
2. It is corrected by the predicted lidar pose efficiently. But you may try to use the predicted IMU pose to have more accurate results.
3. If the value of the next frame is updated by other means, it will not cause a problem. I will dig deep into this later.

hyye, 2019-07-26:
I found you access index greater than the size, which is undefined behavior.

zhaostu4, 2019-07-26:
Sorry, I just set up debugging and found out.

closed 2019-07-26 by zhaostu4