hyye/lio-mapping #30: lidar-imu initialization

llpan91, 2019-08-28:
I confused with the coarse gravity solver，
“    tmp_A = 0.5 * I3x3 * (dt12 * dt12 * dt23 + dt23 * dt23 * dt12);
    tmp_b = (pl2 - pl1) * dt23 - (pl3 - pl2) * dt12
        + (rl2 - rl1) * plb * dt23 - (rl3 - rl2) * plb * dt12
        + rl2 * rlb * dp23 * dt12 + rl1 * rlb * dv12 * dt12 * dt23
        - rl1 * rlb * dp12 * dt23;”
how to construct the least squares method for gravity ？

hyye, 2019-08-28:
You may find some useful information here, https://github.com/hyye/lio-mapping/issues/4#issuecomment-494632602.

narutojxl, 2020-06-15:
Hi @llpan91, Do you figure out what is the residual, when calculating the initial global gravity? Thanks a lot !

goldenminerlmg, 2020-09-07:
@hyye @llpan91 Hi , In the recommend paper ''Monocular Visual–Inertial State Estimation With Online Initialization and Camera–IMU Extrinsic Calibration'' in part V-B, I still confuse how to estimate the gravity? Can you give a detail description for these part:
1)ApproximateGravity(all_laser_transforms, g, transform_lb):
tmp_A = 0.5 * I3x3 * (dt12 * dt12 * dt23 + dt23 * dt23 * dt12);
    tmp_b = (pl2 - pl1) * dt23 - (pl3 - pl2) * dt12
        + (rl2 - rl1) * plb * dt23 - (rl3 - rl2) * plb * dt12
        + rl2 * rlb * dp23 * dt12 + rl1 * rlb * dv12 * dt12 * dt23
        - rl1 * rlb * dp12 * dt23;
2)RefineGravityAccBias(all_laser_transforms, Vs, Bgs, g, transform_lb, R_WI)
      tmp_A.block<3, 3>(0, 0) = dt12 * Matrix3d::Identity();
      tmp_A.block<3, 2>(0, 6) = 0.5 * Matrix3d::Identity() * lxly * dt12 * dt12;
      tmp_b.block<3, 1>(0, 0) = pl2 - pl1 - rl1 * rlb * dp12 - (rl1 - rl2) * plb - 0.5 * g_refined * dt12 * dt12;
      tmp_A.block<3, 3>(3, 0) = Matrix3d::Identity();
      tmp_A.block<3, 3>(3, 3) = -Matrix3d::Identity();
      tmp_A.block<3, 2>(3, 6) = Matrix3d::Identity() * lxly * dt12;
      tmp_b.block<3, 1>(3, 0) = -rl1 * rlb * dv12 - g_refined * dt12;
What's the physical meaning for these fomula? Hope for your answer.
Thank you very much!