hyye/lio-mapping #33: how to measure and set 'msg_time_delay' in the config file?

heyang0105, 2019-09-05:
Hi, I notice that a significant parameter 'msg_time_delay' is explicitly set in the config file. it controls the way to pair the laser data and the imu data. I also found that all '*config.yaml's set it as 0.05, but I don't know why. 
Could you explain more about it, please?  How to set it? How to measure it base the customized devices?

hyye, 2019-09-05:
It is used to buffer more IMU data to the middle timestamp of the lidar (if the frequency of the lidar is 10Hz, with the config set as 0.05).
The lidar points will be transformed by the lidar pose at the middle timestamp later.

heyang0105, 2019-09-10:
Thanks. I appreciate it.
Additionally, is it possible to effectively optimize the dt ( the offset of the imu's and the lidar's sync time)? Like the one Dr.Qin Tong did in his 'Online Temporal Calibration for Monocular Visual-Inertial Systems'. He modeled the feature velocity. However, these features in 'LOAM' are weakly linked by nearest neighbor search.

closed 2019-09-10 by heyang0105