hyye/lio-mapping #36: Discrete time noise covariance

andre-nguyen, 2019-09-13:
Hi,

thank you so much for answering questions :smile:. In VINS-Mono the discrete time noise covariance matrix is approximated to the first order using
[image]

So `dt * Gt * Qt * Gt'` however in your supplementary material your approximation of `Qd` is
[image]

Where does the extra dt come from?

hyye, 2019-09-22:
This is from another convention of discretization.
I discussed with the author of VINS-Mono. The first one is more proper. I will change the supplementary material accordingly.

andre-nguyen, 2019-09-22:
Interesting, do you know the reference? I think you can leave it like that in your supplementary material because that is how you did it in the code.

hyye, 2019-09-23:
For the first one, [this paper](https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=6018993) can be a reference.
For the second one, I cannot find a more official paper.

closed 2019-09-23 by andre-nguyen