hyye/lio-mapping #37: Confused with constraints part

getupgetup, 2019-09-18:
Hi, I'm new in LIO/VIO. According to your paper, I think the most different part between LIO and LOAM is the constraint/factor construction. I found this in your paper:
[image]
In my opinion,  LOAM simply use IMU data to give a predicted pose and focus on the second part of it. LIO add frame-to-frame pose constraint by IMU pre-integration. Am I right?
Threrefore, I'm confused with these constraints. the second part represents point-to-plane residuals and third part represents pose-to-pose residuals. Why can they be "added"? How to determine their covariance? Thank you~

hyye, 2019-09-22:
This is the pre-integration term, as shown in [supplementary material](https://drive.google.com/file/d/1GW4RL8J3tfZWgIuOrBG1CBzat31y1TRQ/view) Eqn. (C.8), not pose-to-pose residuals.

getupgetup, 2019-09-25:
> This is the pre-integration term, as shown in [supplementary material](https://drive.google.com/file/d/1GW4RL8J3tfZWgIuOrBG1CBzat31y1TRQ/view) Eqn. (C.8), not pose-to-pose residuals.

Thank you! Is this part mainly in function: Estimator::SolveOptimization()?

closed 2020-06-13 by getupgetup