hyye/lio-mapping #38: Memery growth...

shishiteng, 2019-09-19:
I've tested lio_estimator_node with my own datasets and found that the memery grew over time. It grew from 9.2M to 2G in about  40 minutes. I think lio_estimator_node is just  a lidar-imu odometry  and it doesn't need to consume so much memory.

hyye, 2019-09-22:
I will investigate this. It might be related to the inherited mapping module.

shishiteng, 2019-09-23:
> I will investigate this. It might be related to the inherited mapping module.

Thanks for your reply, I tried to debug this problem and found that  the memory stopped growing when I commented  on the function _UpdateMapDatabase_ in _Estimator.cc_.

```
        if (!opt_point_coeff_mask_.first())
        {
          // UpdateMapDatabase(opt_corner_stack_.first(),
          //                   opt_surf_stack_.first(),
          //                   opt_valid_idx_.first(),
          //                   opt_transforms_.first(),
          //                   opt_cube_centers_.first());

          DLOG(INFO) << "all_laser_transforms_: " << all_laser_transforms_[estimator_config_.window_size - estimator_config_.opt_window_size].second.transform;
          DLOG(INFO) << "opt_transforms_: " << opt_transforms_.first();
        }
```

But I don't know weather the problem has been solved,I am wondering what is the role of map database in estimator node? In my opnion, there is no need to maintain a map database here. I'm very confused and looking forward to your reply.

shishiteng, 2019-09-23:
> Thanks for your reply, I tried to debug this problem and found that the memory stopped growing when I commented on the function _UpdateMapDatabase_ in _Estimator.cc_.
>
> ```
>         if [...]

And as memory grows processing time of _UpdateMapDatabase_ becomes longer and longer, this may make lio_estimater_node can't run in realtime.

hyye, 2019-09-23:
The mapping module is for the initialization only.
It can be stopped after the initialization.

shishiteng, 2019-09-24:
> The mapping module is for the initialization only.
> It can be stopped after the initialization.

Got it! Thank you.

closed 2019-09-27 by shishiteng

hyye, 2019-10-25:
In a new branch "maplab", not merged and comprehensively tested yet, this should be fixed.