hyye/lio-mapping #45: Achieving real-time performance using VLP-16

TixiaoShan, 2019-11-04:
Thanks for the great work.

I am testing your package and not able to achieve real-time performace on my own dataset (https://github.com/TixiaoShan/Stevens-VLP16-Dataset). I can get at most 3Hz post estimation. I have followed the README and built the package with RELEASE. Can you help take a loop into this problem?

hyye, 2019-11-11:
Hi @TixiaoShan, thanks for testing. I think it may be related to https://github.com/hyye/lio-mapping/issues/20. Replacing the front-end part to your method with fewer features may help. I checked some of your bags, where the IMU frequency is about 70-80 Hz and varying. The IMU quality may also bring some problems. An IMU with a higher frequency >200Hz is recommended.

TixiaoShan, 2019-11-11:
Thanks for the reference. I will take a closer look and report back. @hyye

closed 2019-11-19 by TixiaoShan

mhaboali, 2020-05-17:
Hi @TixiaoShan

I have the same issue, could you help with it?