hyye/lio-mapping #46: Using YVT-35LX

sgk-000, 2019-11-18:
Hi @hyye,
Thank you for releasing this great source code.
I am trying to run lio-mapping with hokuyo3d sensor.
hokuyo3d sensor: https://www.hokuyo-aut.jp/search/single.php?serial=224

However I encountered below error and I don't know How I can resolve the error.
[image]

how would you resolve this error? Also you processed velodyne points(maybe accumulate their points?), How would you adapt lio-mapping to work with hokuyo3d sensor?
  
Thanks,

hyye, 2019-11-21:
Hi @sgk-000, I haven't tested this sensor. My suggestion is to modify the front-end to make the "baseline" method, like "16_scans_test.launch", work first.
The error might be related to invalid lidar inputs or large errors in the state estimation.

sgk-000, 2019-11-23:
Hi @hyye 
Thank you for reply. I tested your code with my sensor and below error message was appeared.

` Failed to find match for field 'ring'.`

Actually our sensor's point cloud doesn't have ring field. I can add it, but I don't know our sensor is good for your code.
Below images shows our sensor's output.

[image]
[image]

Does you code work with only spinning lidar? (It means lidar's output must have ring?).
If so, I want you to suggest how I modify your code. (Actually I haven't understood PointToRing function in PointProcessor.cc)

This algorithm is essential for us. Please your help.
Thank you for your cooperation.

hyye, 2019-11-23:
The ring field is provided by some lidar drivers.
The current front-end for extracting lidar points is only suitable for the spinning lidar. I am not sure if your lidar is similar to DJI's. If yes, you might consider adapting the feature extraction part from [loam_livox](https://github.com/hku-mars/loam_livox/blob/master/source/livox_feature_extractor.hpp) to your application.