hyye/lio-mapping #47: Odom Node Crashes with Ros Melodic

geoeo, 2019-11-26:
Hello,

I successfully compiled the project on my machine, but when I try to run the rosbag the odom node crashes with the following error in the log file.
```

  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/threadpool.py", line 218, in run
    result = cmd(*args)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1316, in single_request
    return self.parse_response(response)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1493, in parse_response
    return u.close()
  File "/usr/lib/python2.7/xmlrpclib.py", line 800, in close
    raise Fault(**self._stack[0])
Fault: <Fault -1: 'publisherUpdate: unknown method name'> 
```

The commands I use to run are the same as in the Readme.md file

geoeo, 2019-11-26:
The issue is not ROS melodic. But rather PCL in release mode. I switched to Debug and it works fine. I think I read about this in the LOAM project issus, where ROS has problems with precompiled PCL. Compiling 1.9.1. locally and linking it could solve this. But I only have 1.8.1 locally which doesnt work with LIO

hyye, 2019-11-26:
Hi @geoeo, I use ROS Melodic with the PCL from apt-get, and it works. Does your log mean that the error is related to the ROS parser or something else? Could you try to run the executives instead of using launch files?

geoeo, 2019-11-26:
I am running lio_processor_node with clion and I get a mem free error related to PCL only in the realease mode. I have tried with apt-get and locally compiled. Apt-get for me was `sudo apt install libpcl-dev
`

geoeo, 2019-11-26:
Ps. You can mitigate the performance impact of Debug by `set(CMAKE_CXX_FLAGS_Debug "-O3 -Wall -g")`

hyye, 2019-11-26:
Thanks for your suggestion on debugging.
I think it is the default `PCL` from `melodic-desktop-full`, since I didn't explicitly install `libpcl-dev` with `apt`. You may check with the [docker](https://github.com/hyye/lio-mapping/blob/master/docker/Dockerfile) example for more details.
I cannot reproduce this issue. Could it be related to the version of `PCL` or some configurations of compiling?

geoeo, 2019-11-26:
docker works fine. I think its compiling configurations of PCL. But im not enough of a ROS/linux expert to understand what is going on.