hyye/lio-mapping #49: Running LIO on KITTI Dataset

geoeo, 2019-11-29:
Hello,
From the commits I can see you have added support for KITTI. But From what I can tell:

-  The tf tree is getting visualized. 

- The LIDAR odometry is working

The IMU based smoothing is not working. How can I get it to work with IMU data? I tried remapping to KITTI topics but no use. Maybe I an looking at the wrong topic. The estimated trajectory should be published on /aft_mapped? correct?

geoeo, 2019-12-05:
It seems to be an issue related to https://github.com/hyye/lio-mapping/issues/47. I cant seem to get the performance with the local version that the docker image has. It works in docker. One last thing: The tf tree that is visualised from the bag file is wrong. I plot the GPS position and it seems to be in-line with the aft_mapped transform

closed 2019-12-05 by geoeo

hyye, 2019-12-19:
I am not sure which transform in the tf tree is wrong. Could it be related to the potential "bug" after initialization?