hyye/lio-mapping #50: rosbag ground thruth not aligned with docker results

geoeo, 2019-12-05:
Hello,

I am plotting the GT pose from the topic /Robot_7/pose with the docker image. However, the aft_mapped/camera and gt_pose transforms are not aligned. Is there some transform step that is not given to align the two?

geoeo, 2019-12-05:
[image]

geoeo, 2019-12-31:
Would it be possible to know what this alignment is? It would make comparison between local and global methods possible

daijiaoshou, 2020-01-01:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    did you know VINS-Fusion, the method to fuse VIO and GPS, maybe we can try to fuse LIO and GPS.
    
        
            
        
    
    
     
    
        
                    
            
                
                    
                            
                                
                            
                            
                                daikai
                            
                    
                        
                            
                                    daikai@jluracing.com
                            
                        
                
            
        
        
    
    
        签名由
        网易邮箱大师
        定制

geoeo, 2020-01-02:
I have only seen the repo. Is there a publication for this method?