hyye/lio-mapping #55: Height drift when using Map Builder

lsdta, 2019-12-19:
Thanks @hyye for the open source code, I've been running your algorithm with my own rosbags, however whenever map builder (4D) is enabled, the trajectory has a slow map height (z-axis) drift, similar to #18.

Disabling 4D (rotational constraints) removes the height drift, however the map produced (3min long dataset) has some errors in Z due to planar motion, just as #18 experienced. The extrinsic estimation step seems to be working correctly. The issue does not occur when using LOAM-only as well, as expected. The rotational constraints refinement sounds promising but I cannot get it to work without this extra height drift.

Has anyone also experienced this problem and managed to fix it? I'm stumped on what is causing this issue.

hyye, 2019-12-19:
Hi @lsdta, thanks for pointing out. My understanding is that the rotational constraints can improve globally orientation error but not the z-axis translation, which may come from the odometry. Disabling this part makes the system almost the same as the LOAM's mapping. BTW, setting `skip_count` in the `mapper` to `1` may reduce the drift from the odometry by avoiding taking the odometry directly.

lsdta, 2019-12-19:
Thank you for the quick reply. Am I right in that setting _enable_4D_=false and running processor/estimator/mapbuilder nodes is essentially running LIO (rather than LIO-mapping)? Or does _enable_4D=false revert it back to your implementation of LOAM?

I will try out the rotational constraint with _skip_count_=1, is there any reason other than computational efficiency that pose optimization is not run every step of odometry by default?

lsdta, 2019-12-19:
> 
> 
> Am I right in that setting `enable_4D=false` and running processor/estimator/mapbuilder nodes is essentially running LIO (rather than LIO-mapping)? Or does `enable_4D=false` revert it back to your implementation of LOAM?
> 

And if it does revert it to your implementation of LOAM, which part of the code needs removing to run LIO rather than LIO-mapping (as designated in your paper). That is, how to run LIO without rotational constraints but continue using IMU updates?

hyye, 2019-12-19:
> Or does `enable_4D=false` revert it back to your implementation of LOAM?

Yes. It will still take IMU information in the odometry node (two separated nodes).
> I will try out the rotational constraint with `skip_count=1`, is there any reason other than computational efficiency that pose optimization is not run every step of odometry by default?

The not for efficiency, the computation time remains almost the same. It's for some tests.

lsdta, 2019-12-19:
> Yes. It will still take IMU information in the odometry node (two separated nodes).

I am finding that `/predict_odom` can often deviate quite far from `/aft_mapped_to_init` when `enable_4D` is off, which is what led me to believe that it was not using the IMU data at all for odometry.

I also find that the IMU noise parameters affect odometry significantly, have you experimented with these values? (given that they are from VINS-mono). I do not fully understand their implementation (as to what units they are using in their model) but lowering the values seems to give more weighting to the IMU for odometry, and vice-versa.

lsdta, 2019-12-20:
@hyye I also noticed that your video only showed Top-Down view when doing the car-tests, did you also experience error in Z-axis when doing planar-motion?

hyye, 2019-12-20:
> I am finding that `/predict_odom` can often deviate quite far from `/aft_mapped_to_init` when `enable_4D` is off, which is what led me to believe that it was not using the IMU data at all for odometry.

It does affect the `mapper` node only.

> I also noticed that your video only showed Top-Down view when doing the car-tests, did you also experience error in Z-axis when doing planar-motion?

There is no z-axis drift in the golf-cart tests in my video.

lsdta, 2019-12-23:
Thank you, you have cleared some things up and I believe height drift is coming from an incorrect odometry that is only mapped incorrectly when using 4D.

closed 2019-12-23 by lsdta

LarryDong, 2024-12-11:
I have the same issue. 
In my own simulation dataset, some sequences has extremly huge z-axis drift, but some are not.
disable `enable_4D` reduce the drift.
Still not know why and how to solve this issue.