hyye/lio-mapping #57: try it on track robot！

njauwangkai, 2020-01-12:
First of all, thank you for your contribution, I still encountered a little problem, and it worked very well when I was holding the map.like this  
[image] 
Calibration parameter selection 1 Roughly measured external parameters of inertial navigation and radar. But I prevent radar and inertial navigation on tracked robots. At the time of initialization, I have controlled the robot to make a large range of steering, but there are still problems with the mapping effect. There is no IMU incentive, but the map is obviously misaligned!

[image]

[image]
May I ask if I can hold the sensor for initialization and get its more accurate external parameters, and then write the external parameters into the YML file. Will the effect be better during initialization?
Thank you friends looking forward to your reply

hyye, 2020-02-13:
> May I ask if I can hold the sensor for initialization and get its more accurate external parameters, and then write the external parameters into the YML file.

This is a way to determine the external parameters.

As I suggested in https://github.com/hyye/lio-mapping/issues/53#issuecomment-567310010, the initialization part for some planar motion vehicles can be modified to a static version.
I just implemented a very simple static initializer in a new branch [static_init](https://github.com/hyye/lio-mapping/tree/static_init).
Adding
```yaml
static_init: 1
```
to the previous config file will enable the static initializer of IMU ([example](https://github.com/hyye/lio-mapping/blob/83a0500a776d6e036a47c9866e2a3883f6fdba57/config/bpearl_top.yaml#L73)).

njauwangkai, 2020-02-16:
thank you very  much!   I will try it when    thing  be ok !

closed 2020-03-10 by njauwangkai