hyye/lio-mapping #62: Assertion `point_representation_->isValid (point) && "Invalid (NaN, Inf) point coordinates given to nearestKSearch!"' failed.

zhangzhouyuan, 2020-04-16:
Hi，
I try to use lio_estimator_node to generate odom，But an error occurred

 roslaunch lio test_indoor.launch 
... logging to /home/zyx/.ros/log/31290dc8-7faa-11ea-aefb-e8d8d16fbe3d/roslaunch-zyx-HP-Laptop-14s-cr2xxx-28107.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zyx-HP-Laptop-14s-cr2xxx:45375/

SUMMARY
========

PARAMETERS
 * /lio_estimator/config_file: /home/zyx/test/sr...
 * /lio_processor/infer_start_ori: False
 * /lio_processor/rad_diff: 1.0
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    lio_estimator (lio/lio_estimator_node)
    lio_processor (lio/lio_processor_node)

ROS_MASTER_URI=http://localhost:11311

process[lio_processor-1]: started with pid [28145]
process[lio_estimator-2]: started with pid [28146]
I0416 14:19:16.431304 28145 PointProcessor.cc:82] lower_bound: -15
I0416 14:19:16.431484 28145 PointProcessor.cc:83] upper_bound: 15
I0416 14:19:16.431505 28145 YamlLoader.cc:13] config_file: 
I0416 14:19:16.431526 28145 processor_node.cc:90] laser_scans: 16
I0416 14:19:16.485013 28146 YamlLoader.cc:13] config_file: /home/zyx/test/src/LIO_mapping/lio-mapping/config/indoor_test_config.yaml
I0416 14:19:16.493486 28146 MeasurementManager.cc:50] msg_time_delay: 0.05s
I0416 14:19:16.493522 28146 MeasurementManager.cc:51] scan_period: 0.1s
I0416 14:19:27.111759 28158 Estimator.cc:544] >>>>>>> extrinsic calibration failed <<<<<<<
I0416 14:19:27.902637 28158 Estimator.cc:544] >>>>>>> extrinsic calibration failed <<<<<<<
I0416 14:19:28.699381 28158 Estimator.cc:544] >>>>>>> extrinsic calibration failed <<<<<<<
I0416 14:19:29.526885 28158 Estimator.cc:544] >>>>>>> extrinsic calibration failed <<<<<<<
I0416 14:19:30.313222 28158 Estimator.cc:544] >>>>>>> extrinsic calibration failed <<<<<<<
I0416 14:19:31.143107 28158 Estimator.cc:544] >>>>>>> extrinsic calibration failed <<<<<<<
I0416 14:19:31.929845 28158 Estimator.cc:544] >>>>>>> extrinsic calibration failed <<<<<<<
I0416 14:19:32.721494 28158 Estimator.cc:544] >>>>>>> extrinsic calibration failed <<<<<<<
I0416 14:19:33.516692 28158 Estimator.cc:544] >>>>>>> extrinsic calibration failed <<<<<<<
I0416 14:19:34.312844 28158 Estimator.cc:544] >>>>>>> extrinsic calibration failed <<<<<<<
I0416 14:19:35.121338 28158 Estimator.cc:544] >>>>>>> extrinsic calibration failed <<<<<<<
I0416 14:19:35.940508 28158 Estimator.cc:544] >>>>>>> extrinsic calibration failed <<<<<<<
I0416 14:19:36.727543 28158 Estimator.cc:544] >>>>>>> extrinsic calibration failed <<<<<<<
I0416 14:19:37.529842 28158 Estimator.cc:544] >>>>>>> extrinsic calibration successful <<<<<<<
[ WARN] [1587017977.551398513]: >>>>>>> IMU initialized <<<<<<<
lio_estimator_node: /build/pcl-OilVEB/pcl-1.8.1+dfsg1/kdtree/include/pcl/kdtree/impl/kdtree_flann.hpp:136: int pcl::KdTreeFLANN<PointT, Dist>::nearestKSearch(const PointT&, int, std::vector<int>&, std::vector<float>&) const [with PointT = pcl::PointXYZI; Dist = flann::L2_Simple<float>]: Assertion `point_representation_->isValid (point) && "Invalid (NaN, Inf) point coordinates given to nearestKSearch!"' failed.
[lio_estimator-2] process has died [pid 28146, exit code -6, cmd /home/zyx/test/devel/lib/lio/lio_estimator_node __name:=lio_estimator __log:=/home/zyx/.ros/log/31290dc8-7faa-11ea-aefb-e8d8d16fbe3d/lio_estimator-2.log].
log file: /home/zyx/.ros/log/31290dc8-7faa-11ea-aefb-e8d8d16fbe3d/lio_estimator-2*.log

How can I solve this problem？
Thanks.

EdCheriee, 2020-06-18:
I have exactly the same issue when I run an outdoor_64 launch file with my personal recorded data. Have you managed to find a solution to this or the reason behind it?

EdCheriee, 2020-06-18:
Hey, so the way I managed to solve this is to put correct R and T matrices for the config, after that calibration is succesfull and I do not get this error. Hope this helps you.

shaheer34mts, 2021-12-13:
Hi, i am having the similar issue. My rosbag is around 630 seconds large, and it fails around 343 seconds marks giving the same error as mentioned in this post. No matter what I values I put in the extrinsic matrices, it always fails at the same point. any suggestions ?