hyye/lio-mapping #66: Trying to understand how the optoimization parameters work for Lio-mapping

SoguMax, 2020-05-25:
Hey. My question is are follows. What do the following parameters do in the indoor_test_config.yaml file in the lio_mapping package?

# optimization options
run_optimization: 1

update_laser_imu: 1
gravity_fix: 1

plane_projection_factor: 0

imu_factor: 1
point_distance_factor: 1

prior_factor: 0

marginalization_factor: 1

odom_io: 6

pcl_viewer: 0

I am especially interested in odom_io as it seems to be affecting the trajectory of my odometry, but I am not sure how it affects it.

Thanks

[image]

hyye, 2020-05-26:
Hi @SoguMax, thanks for your interest.
For some of the parameters you may refer to https://github.com/hyye/lio-mapping/issues/20#issuecomment-514591841.
The `odom_io` is used to skip several frames. For example, `odom_io: 2` means that it will take 1 point cloud from every **2** point clouds.

SoguMax, 2020-05-26:
> For some of the parameters you may refer to #20 (comment)

Thank you so much for your help.

SoguMax, 2020-05-26:
Hi, Hoayang

There seems to be a drift in the z direction  when  moving and capturing
hrough several floors in an indoor environment. What do you recommend?
Moving the lidar and imu through a faster motion (The bag file is recorded
with a slow walking pace) or just  adjusting this odom_io parameter.

Thank you

Samuel

kissb2, 2020-06-18:
@SoguMax 
similar issue. 
I'm testing in an outdoor environment
also z drift occurs,   
any advices?

hyye, 2020-06-19:
Hi @SoguMax and @kissb2, thanks for your interest. As I mentioned in https://github.com/hyye/lio-mapping/issues/18#issuecomment-513747567, this drift is inevitable if the path is long and of no local loop.  Adding a loop detection module, e.g., [segmap](https://github.com/ethz-asl/segmap), and optimize the pose graph may help.