hyye/lio-mapping #67: can't access the Supplementary Material in reference[12]

xieqi1, 2020-06-01:
The Supplementary Material which in https://sites.google.com/view/lio-mapping can't be accessible, I can't wait to read this material to help me better understand this excellent work. so is there any other way to get it? Thank you!

DarrenWong, 2020-06-07:
seems direct pdf cannot be uploaded, unzip the file you can view the supplementary materials.  你需要一把梯子...

[Supplementary Material to-Tightly Coupled 3D Lidar Inertial Odometry and Mapping.pdf.zip](https://github.com/hyye/lio-mapping/files/4741257/Supplementary.Material.to-Tightly.Coupled.3D.Lidar.Inertial.Odometry.and.Mapping.pdf.zip)

xieqi1, 2020-06-07:
get it ! Thank you very much ! 笔芯

closed 2020-06-07 by xieqi1

xieqi1, 2020-06-10:
it seems the link can be directly accessible today~