hyye/lio-mapping #69: q_L and q_R typo in math_utils.h ?

narutojxl, 2020-06-19:
Hi @hyye thanks for your excellent work :)
* The rotation convention in your code is Hamilton, correct? 
*  q_L and q_R matrix in math_utils.h is JPL?
Thanks for your help!

hyye, 2020-06-19:
Hi @narutojxl,
Thanks for your question. The conversion is all in Hamilton as Eigen's.

narutojxl, 2020-06-19:
In [joan's paper](https://arxiv.org/pdf/1711.02508.pdf), his rotation is also Hamilton.
[image]
In math_utils.h
>
	template<typename Derived>
	inline Eigen::Matrix<typename Derived::Scalar, 4, 4> LeftQuatMatrix(const Eigen::QuaternionBase<Derived> &q) {
	  Eigen::Matrix<typename Derived::Scalar, 4, 4> m;
	  Eigen::Matrix<typename Derived::Scalar, 3, 1> vq = q.vec();
	  typename Derived::Scalar q4 = q.w();
	  m.block(0, 0, 3, 3) << q4 * I3x3 + SkewSymmetric(vq);
	  m.block(3, 0, 1, 3) << -vq.transpose();
	  m.block(0, 3, 3, 1) << vq;
	  m(3, 3) = q4;
	  return m;
	}
The code doesn't consistent with above definition?

hyye, 2020-06-19:
I think it is exactly the same, except the order of the quaternion, right?

narutojxl, 2020-06-20:
`q_L in code is: `
[image]
`joan q_L is:`
[image]
I don't understand why they are the same, right? :)

hyye, 2020-06-20:
Since the orders of the coefficients are different. If you swap `q_v` and `q_w`, these two matrices are the same, right???

narutojxl, 2020-06-22:
Hi @hyye, thanks for your patience with me :), thanks a lot ! 
* I see the eigen doc about quaternion's `vec()` function. It returns the imaginary part(q_x, q_y, q_z), matching joan's quaternion definition(q_w is scalar part, [q_x, q_y, q_z] is imaginary part). 
*  `x(), y(), z()` return the x, y, z coefficient respectively of quaternion. `w()` return the w coefficient.
[image]

closed 2020-06-23 by hyye