hyye/lio-mapping #79: Error when using imu data

he-guo, 2020-11-16:
hello,When I use imu data, the following error will be reported
    E1116 11:22:05.977866 27339 MeasurementManager.cc:112] imu message in disorder!
    Of course, the imu data is synthesized, not the original data, because the use of the original imu data will cause the map to be disordered due to severe jitter

vicententu, 2020-12-15:
I have the same issue, have you found any solution to this? thank you.

he-guo, 2020-12-15:
sorry, I haven't solved it yet. The original IMU data will not report this error. I think the IMU data I processed is the reason for the error

he-guo, 2020-12-15:
> I have the same issue, have you found any solution to this? thank you.

sorry, I haven't solved it yet. The original IMU data will not report this error. I think the IMU data I processed is the reason for the error