hyye/lio-mapping #84: catkin_make fails on 98%

baberjunaid, 2022-04-25:
Hello experts, 

All installation went fine. But it get failed on 98% with following error

```
[ 98%] Built target os_cloud_node
/usr/bin/ld: /home/robair/RobAIR/catkin_ws/devel/lib/lio/lio_estimator_node: hidden symbol `_ZN5ceres12CostFunctionD2Ev' in /usr/local/lib/libceres.a(cost_function.cc.o) is referenced by DSO
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [lio-mapping/CMakeFiles/lio_estimator_node.dir/build.make:456: /home/robair/RobAIR/catkin_ws/devel/lib/lio/lio_estimator_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:7251: lio-mapping/CMakeFiles/lio_estimator_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j4 -l4" failed
```

Hi-Zed, 2023-03-09:
I am having the same issue on Ubuntu 20.04.
Did you find a solution?

Hi-Zed, 2023-03-10:
For anyone with a similar issue, I solved it.
You need to build Cerse with shared libs, which are disabled by default.
`option(BUILD_SHARED_LIBS "Build Ceres as a shared library." ON)`

hankge, 2023-07-22:
Hi @Hi-Zed, Where is this sentence you mentioned? How can I change or add this?